bibigone/k4a.net #2: Unity3d Support

Itpyc, 2019-07-25:
Dear Bibigone,When will your work Support Unity3d?Do you have the demo now?thanks!

Pycorax, 2019-07-28:
To add on to this, I was able to get color images to display on a texture on an object in Unity. However, initializing the camera with a DeviceConfiguration that includes setting the DepthMode will always raise a exception:

**Device Configuration:**
```
DeviceConfiguration config = DeviceConfiguration.DisableAll;
config.CameraFps = FrameRate.Thirty;
config.ColorFormat = ImageFormat.ColorMjpg;
config.ColorResolution = ColorResolution.R1080p;
config.DepthMode = DepthMode.NarrowViewUnbinned;
config.SynchronizedImagesOnly = true;
```
**Exception:**
```
ArgumentException: Invalid value of one of the following arguments: config
K4AdotNet.Sensor.Device.ThrowException (System.String[] argNames) (at <0c74825c19ee47d5b1dc0c140254537d>:0)
K4AdotNet.Sensor.Device.CheckResult (K4AdotNet.NativeCallResults.Result result, System.String[] argNames) (at <0c74825c19ee47d5b1dc0c140254537d>:0)
K4AdotNet.Sensor.Device.StartCameras (K4AdotNet.Sensor.DeviceConfiguration config) (at <0c74825c19ee47d5b1dc0c140254537d>:0)
```

The same code on a standard .NET Core 2.2 project in Visual Studio has no such issues running the same code.

Pycorax, 2019-07-29:
I managed to fix the issue based on this issue filed on the main repo here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/484

However, now I'm running into an issue where Unity is unable to find k4abt.dll for some reason.

bibigone, 2019-07-30:
@Pycorax

> However, now I'm running into an issue where Unity is unable to find k4abt.dll for some reason.

Did you try the latest version?

Itpyc, 2019-08-01:
In Unity3d, I cannot use dnn_model.onnx .Do you have the solution?Thanks

bibigone, 2019-08-01:
Body tracking runtime loads `dnn_model.onnx` from current directory. For this reason I'm changing current directory while calling native function to create tracker object. See implementation of method `Sdk.TryCreateTrackerHandle` for details. But I didn't test this solution under Unity3D (I've tested it only with .NET Core and .NET Framework).

Itpyc, 2019-08-01:
> Body tracking runtime loads `dnn_model.onnx` from current directory. For this reason I'm changing current directory while calling native function to create tracker object. See implementation of [...]

Dear,thanks for your reply.I get a problem when try to run the programe.

> “K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: DefaultDomain): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: DefaultDomain): 已加载“F:\sdk\CSKA4\k4a.net-master\K4AdotNet.Samples.Wpf.BodyTracker\bin\x64\Debug\K4AdotNet.Samples.Wpf.BodyTracker.exe”。已加载符号。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“F:\sdk\CSKA4\k4a.net-master\K4AdotNet.Samples.Wpf.BodyTracker\bin\x64\Debug\K4AdotNet.Samples.Wpf.Common.dll”。已加载符号。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\PresentationFramework\v4.0_4.0.0.0__31bf3856ad364e35\PresentationFramework.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\WindowsBase\v4.0_4.0.0.0__31bf3856ad364e35\WindowsBase.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_64\PresentationCore\v4.0_4.0.0.0__31bf3856ad364e35\PresentationCore.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Xaml\v4.0_4.0.0.0__b77a5c561934e089\System.Xaml.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“D:\VISUAL STUDIO 2019\Common7\IDE\Remote Debugger\x64\Runtime\Microsoft.VisualStudio.Debugger.Runtime.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Configuration\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Configuration.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Xml\v4.0_4.0.0.0__b77a5c561934e089\System.Xml.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\mscorlib.resources\v4.0_4.0.0.0_zh-Hans_b77a5c561934e089\mscorlib.resources.dll”。模块已生成，不包含符号。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“F:\sdk\CSKA4\k4a.net-master\K4AdotNet.Samples.Wpf.BodyTracker\bin\x64\Debug\K4AdotNet.dll”。已加载符号。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\PresentationFramework.Aero2\v4.0_4.0.0.0__31bf3856ad364e35\PresentationFramework.Aero2.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\PresentationCore.resources\v4.0_4.0.0.0_zh-Hans_31bf3856ad364e35\PresentationCore.resources.dll”。模块已生成，不包含符号。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Windows\Microsoft.Net\assembly\GAC_MSIL\PresentationFramework-SystemXml\v4.0_4.0.0.0__b77a5c561934e089\PresentationFramework-SystemXml.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
“K4AdotNet.Samples.Wpf.BodyTracker.exe”(CLR v4.0.30319: K4AdotNet.Samples.Wpf.BodyTracker.exe): 已加载“C:\Users\lenovo\AppData\Local\Temp\VisualStudio.XamlDiagnostics.21960\Microsoft.VisualStudio.DesignTools.WpfTap.dll”。已跳过加载符号。模块进行了优化，并且调试器选项“仅我的代码”已启用。
程序“[21960] K4AdotNet.Samples.Wpf.BodyTracker.exe”已退出，返回值为 -1066598274 (0xc06d007e) 'Module not found'。

**Besides,I am a newer with wpf,your samples are too difficult for me to understand.Can you provide me or us a simple sample about body track when you are free. thank you  very much, your work is so helpful,thanks for your great job,**

bibigone, 2019-08-01:
Dear @Itpyc, please provide me with exception (message and call stack) rather than debug output.

Itpyc, 2019-08-04:
I test it in unity3d,work fine.thank you

closed 2019-08-04 by Itpyc