bibigone/k4a.net #4: Green Screen sample

JpEncausse, 2019-08-04:
Hello,
The installation works very well comapred to official SDK that have a lot of link error. Thanks !

Could you provide a sample to perform a "green screen" effect ? In Nutshell it's displaying all the color pixel between a given depth range.

I dig a little bit in the code but you directly call 'C' code for transformation. Is there a sample where you work on the depth array or color array (may be it require OpenCV for perf) ?

bibigone, 2019-08-05:
Hello @JpEncausse,

> Could you provide a sample to perform a "green screen" effect ? In Nutshell it's displaying all the color pixel between a given depth range.

Good idea for sample. I'll definitely implement it soon.

> I dig a little bit in the code but you directly call 'C' code for transformation. Is there a sample where you work on the depth array or color array (may be it require OpenCV for perf) ?

Some code example you're asking about can be found in project `K4AdotNet.Samples.Wpf.BodyTracker` - see class [`ImageVisualizer.ColorBgraImageVisualizer`](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Wpf.BodyTracker/ImageVisualizer.cs) lines 182-211

baSSiLL, 2019-08-27:
Hi @JpEncausse 
Check out a new WPF sample named **BackgroundRemover**.

closed 2019-09-03 by baSSiLL