bibigone/k4a.net #6: Unity setup sample

feiou, 2019-08-09:
Hello,

Thank you for making this repo. It is relatively the best C# Azure Kinect SDK in GitHub so for.

As you've stated this is compatible with Unity 2018 and up, but I think it is not clear how exactly it should be set up to work. I've tried moving the folder into a Unity project or build the source into dll in VS then put into Unity but it still has some issue that it would not work. And plus the mess on the k4a and k4abt dependencies.

It would be best to have a step by step setup to make this work in Unity, or a sample Unity project?

Thanks

bibigone, 2019-08-11:
Hello @feiou,

@baSSiLL is working on Unity sample for this library and I've just merged his first pull request to main branch. Please, take a look: https://github.com/bibigone/k4a.net/tree/master/K4AdotNet.Samples.Unity

qazqaz12378, 2019-08-21:
hello
Run the error after compiling according to the tutorial you said.
[image]
Thanks

baSSiLL, 2019-08-21:
Hi @qazqaz12378 
Is **depthengine_1_0.dll** contained in the root directory of the sample Unity project? It should have been copied there by **prepare.cmd** script when you run it.

qazqaz12378, 2019-08-22:
@baSSiLL  My problem, I can copy the dll all over and run normally, thank you

closed 2019-08-22 by baSSiLL