bibigone/k4a.net #8: support for UWP?

guidout, 2019-08-21:
hi, 
thank you for this. This is awesome. 
Since this supports Core .NET, does it work for UWP development?
I tried to install the NuGet but it doesn't seem to load properly.

Thanks,
Guido

bibigone, 2019-08-21:
This library depends on native DLLs from Azure Kinect Sensor SDK. And unfortunately there is no version of these native DLLs for UWP. You can upvote appropriate feature request:
https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/37615828-uwp-support

guidout, 2019-08-21:
gotcha. I'm surprised tho. I'm no expert, but I'd think that if it works for a .NET Core console app it would also for UWP. They are both based on .NET Core.
Anyhow, great work. Thank you!

closed 2024-01-21 by bibigone