bibigone/k4a.net #9: Support for 0.9.2

hmexx, 2019-09-02:
Hey dude.

Amazing work. Your repo just makes getting started with the Kinect and C# a million times easier than the official repo.

One thing though. Do you have any idea what mods it needs to work with 0.9.2?

It only came out recently so I'm not surprised it doesn't work with it, but apparently they've improved the tracking model, both in accuracy and speed.

The MSI is here:
https://www.microsoft.com/en-us/download/details.aspx?id=100128

bibigone, 2019-09-04:
Hello @hmexx,

Thank you for your feedback. Please take a look at commit 7ce55d6

hmexx, 2019-09-04:
Perfect! Cloning it now. Looks like you had to change quite a bit of code so I'm glad I didn't attempt it myself. Thanks again for your great work. I've already started building a proof of concept on top of it for a project I'm working on.

closed 2019-09-09 by bibigone