bibigone/k4a.net #10: depthimage Stream demo

jimzou, 2019-09-03:
Thank you very much for your amazing work. Can you share an examples of depth image stream demo in unity3D?

baSSiLL, 2019-09-03:
Hi! Do you mean rendering depth frames to a texture similarly to how the color stream is rendered in provided Unity sample? Or you want to render 3D point cloud obtained from depth data?

jimzou, 2019-09-03:
> Hi! Do you mean rendering depth frames to a texture similarly to how the color stream is rendered in provided Unity sample? Or you want to render 3D point cloud obtained from depth data?

I want to render depth frames to a texture！

baSSiLL, 2019-09-03:
We don't have a ready code for this.
The difference from rendering a color stream is that you cannot just take the raw data and copy it into a texture to get a color image. You need to transform pixels from 16-bit depth values in millimeters to colors (usually using some palette). You can find an example of such transformation made on CPU in [DepthImageVisualizer](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Wpf.Viewer/ImageVisualizer.cs) class. Note that you'll need to allow unsafe code in Player Settings to use that code in Unity scripts.

However, if you're looking to implement this in production then it's better to perform depth->color transformation with a custom shader. I suppose you can use **Graphics.Blit** method for this, with depth data loaded into source texture in R16 format, and custom shader assigned to a material passed to the method.

baSSiLL, 2019-09-03:
@jimzou Check out updated Unity sample. I've implemented rendering of the depth stream using shader-based approach outlined above. The most important are new [DepthStreamRenderer](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Unity/Assets/Scripts/DepthStreamRenderer.cs) class and [DepthToColor](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Unity/Assets/Shaders/DepthToColor.shader) shader.

jimzou, 2019-09-04:
Thank you very much for your work.

closed 2019-09-04 by baSSiLL