bibigone/k4a.net #11: Unity Sample with avatar

Itpyc, 2019-09-05:
Thanks for your Great Job. I want to animate an avatar in Unity by mapping the user's skeleton and model skeleton, Can you kindly give me some ideas or provide  a sample. @baSSiLL Thanks very much.

baSSiLL, 2019-09-13:
Hi @Itpyc 
I've added an implementation of motion transfer to a character based on joint rotations. See it in action by switching to "Character" mode when running the sample app. Main code is contained in [CharacterAnimator](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Unity/Assets/Scripts/CharacterAnimator.cs) component.
Note that this implementation is pretty naive in the sense it does not employ any techniques to improve the look of resulting motion - like tweaking rotations based on differences in proportions between actor and character or preventing foot slide.

I have plans to add another implementation based on Unity IK solver and positions of limb endings. But I'm not sure how soon I'll be able to work on this.

Itpyc, 2019-09-16:
> Hi @Itpyc
> I've added an implementation of motion transfer to a character based on joint rotations. See it in action by switching to "Character" mode when running the sample app. Main code is [...]

That's all I need. Thank you very much. @baSSiLL

closed 2019-09-16 by Itpyc