bibigone/k4a.net #12: Unity Body Tracking is not working

tomriddle7, 2019-09-24:
I copied libraries and data files for Body Tracking to \k4a.net-master\K4AdotNet.Samples.Unity\Assets\Plugins\K4AdotNet but I got error that 'Cannot initialize body tracking: Cannot find Body Tracking runtime or some of its components (neither in application directory, nor in Body Tracking SDK directory).'
Why this happend?
p.s. I modified 166 line in RecordThenPlaybackTest.cs cause build error. Is that reason?

bibigone, 2019-09-24:
Hello @tomriddle7,

> I copied libraries and data files for Body Tracking to \k4a.net-master\K4AdotNet.Samples.Unity\Assets\Plugins\K4AdotNet

What files and what version did you copied?

The latest version is compatible with Body Tracking **0.9.3** and the following files are required:
* k4abt.dll
* dnn_model_2_0.onnx
* cudnn64_7.dll
* cublas64_100.dll
* cudart64_100.dll,
* vcomp140.dll

> Why this happend?

The easiest way is to install Body Tracking SDK **0.9.3** from here: https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download (MSI) to default destination folder.

> p.s. I modified 166 line in RecordThenPlaybackTest.cs cause build error.

Thank you for letting us know. We fixed this string literal in the latest commit.

> Is that reason?

Definitely not )

tomriddle7, 2019-09-24:
It works well! Thanks!

closed 2019-09-24 by tomriddle7