bibigone/k4a.net #14: Doesn't seem to support multiple devices syncing? or I am doing it wrong?

feiou, 2019-09-25:
Trying to do 2 or more kinects together synced on one computer in Unity to drive interaction of the visuals using your library. Worked like a charm with one Kinect, but could not make it work with 2 or more synced with audio cable.

bibigone, 2019-09-25:
To use synchronization you have to:
1. Connect audio cable to SYNC OUT jack of master device and set `DeviceConfiguration.WiredSyncMode` to `WiredSyncMode.Master` value for this device in `Device.StartCameras` call
2. Also, you have to start color stream for master device even if you're not going to use color data. This is because color camera is used to generate sync impulses.
3. Connect audio cable to SYNC IN jack of another device and set `DeviceConfiguration.WiredSyncMode` to `WiredSyncMode.Subordinate` value for this device in `Device.StartCameras` call.
4. It is also recommended to set `DeviceConfiguration.SubordinateDelayOffMaster` to 160 usec or more for subordinate device to minimize mutual devices interference (see https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/436).

feiou, 2019-09-25:
Hello bibigone,

Thanks for the fast reply.

We managed to get 2 Kinects working separately as standalone. But could not get them to sync as master and subordinate. Our step is the same as the step you mentioned. 

But whenever we set " WiredSyncMode = WiredSyncMode.Master" in the configuration, we have this error:

[image]

Another confusion is: do we need to connect the sub Kinect to the computer with usb-c too? On the official guide they say only connect master to the computer, but it hasn't worked for me yet. I connect both to the computer for the 2 Kinect standalone setup.

Moreover, just want to make sure, is there any way to use 2 Kinect also for Body Tracking?

Thanks! Your help is super appreciated!

bibigone, 2019-09-25:
> But whenever we set " WiredSyncMode = WiredSyncMode.Master" in the configuration, we have this error

Hmm. Did you connect 3.5 audio jack to SYNC **OUT** before this call? If not, please connect it. If yes, then please check value of `Device.IsSyncOutConnected` property before calling of `Device.StartCameras` - it must return `true`. If it returns `false`, then something is wrong with your connection to SYNC OUT.

> Another confusion is: do we need to connect the sub Kinect to the computer with usb-c too?

Sure! Audio cable is used only for shutter sync impulses and cannot transfer color/depth streams.  But you can connect second Kinect device to different PC if needed.

> Moreover, just want to make sure, is there any way to use 2 Kinect also for Body Tracking?

Current version of Body Tracking SDK supports only single sensor. See https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/519#issuecomment-514308292

feiou, 2019-09-25:
Thanks man! Now it syncs with each other, it was a bad audio cable.. What are the odds... 

But as it is working, here rises another question, what's good for syncing devices together? Cause I did a side-by-side to compare 2 standalone kinect and 2 synced kinect and I couldn't really tell the difference.

Thanks for your help!

closed 2019-10-01 by bibigone

bibigone, 2019-10-01:
> what's good for syncing devices together?

It can can be useful to track fast motions and/or fast moving objects from different points of view. Also, it is good for some photogrammetry tasks.

> Cause I did a side-by-side to compare 2 standalone kinect and 2 synced kinect and I couldn't really tell the difference.

To see the difference you have to do frame-by-frame comparison on some fast motions.

In any case, if you don't see any advantages in using of external sync for your applications then simply don't use it )