bibigone/k4a.net #17: Possition tracking of 3d object

remi0s, 2019-10-13:
Is it possible with the current library to keep track of the possition of an object? 
I need to keep track of a gun's replica in order to use it inside a unity game.  I was thinking using the body tracking library in order to attach the weapon on the hand joints but this is not optimal as the replica might get infront of the line of site of the hands while also it will always have to be on model character.  So i would like to know if it is possible to keep track of the replica it self. 
How can i manage this task? Does your library support object tracking beside the body tracking? Any instructions of how to get started or example would be really appreciated. 
Thank you for your time and for sharing this amazing library.

baSSiLL, 2019-10-14:
k4a.net does not implement any tracking algorithms on its own. It only provides a .NET wrapper for capabilities present in Microsoft SDKs for Azure Kinect. And at the moment there is such feature in SDKs.
Speaking of a probable solution to your task - this depends on your requirements for accuracy, allowed positions/motions of a player, whether you control/manufacture gun replicas etc. For instance, if you could paint ends of a gun with distinct colors you might look for those colors in camera's color picture and then determine corresponding depth and 3D coords.

remi0s, 2019-10-15:
Thank you for your answer. Can you point me towards an example? Do i need to use an opencv library in unity for this kind of task? Also i think infrared tracking could be a valiable solution but in general i find hard to find examples for the azure kinect dk.

baSSiLL, 2019-10-15:
Sorry, I cannot point you to any ready-made example.

closed 2019-10-16 by bibigone