bibigone/k4a.net #18: [Unity] Support multiple body tracking and multiple devices in next version?

gitldg, 2019-10-22:
First of all, thank you very much for sharing. This library is very helpful for my current unity project. In my work, I need to use multiple azure kinect dk devices to track multiple people's actions.

However, the example scene is not perfect, and now it can only be tracked in one human body. When there are more than one person, there will be a frame skipping situation. It is expected that the next version will be able to add multiple body tracking function.

In addition, there may be an excessive expectation: the new function of using multiple devices to track multiple people.

Thanks again for sharing.

gitldg, 2019-10-23:
After trying, one AK device can track multiple human bodies and work normally in Unity.

The solution is to add a **bodyID** attribute to Skeleton struck(**at the end** of Skeleton.cs file):
    1. [SkeletonProvider.cs] bodyID value of Skeleton is assigned with bodyFrame.GetBodyId(i) in Update().
    2. [SkeletonRenderer.cs] the SkeletonProvider_SkeletonUpdated() function is modified to:
    
       private void SkeletonProvider_SkeletonUpdated(object sender, SkeletonEventArgs e)
            {
                if (e.Skeleton == null)
                {
                    HideSkeleton();
                }
                else
                {
                    if (e.Skeleton.Value.bodyID != bodyID)
                        return;
                    RenderSkeleton(e.Skeleton.Value);
                }
            }

baSSiLL, 2019-10-23:
Hi @gitldg 
I'm glad to see you had some progress in your task.
We're not aimed yet to bake a comprehensive script library for Unity. The sample is there just to show how K4A.Net library can be incorporated into a Unity project.
K4A.Net itself provides most features of SDKs for Azure Kinect including support for multiple cameras and tracking of multiple persons (individually by each camera).

gitldg, 2019-10-24:
@baSSiLL 
Thank you for your reply.

After another attempt, the function of multi device tracking multi human body can work.

Before, I tried to connect two AK devices to the same PC and transfer the capture data of the two AK devices into the same tracker object, resulting in a half reduction in the number of bodyframe data obtained, causing an exception.
When i try to solve the problem by creating multiple tracker objects, i get an error prompt:"Oops! Current version of Body Tracking runtime does not support creation of multiple body trackers. Sorry!"

Finally, I connect each AK device to a separate PC and it works.

Thank you, dear Russian brother

remi0s, 2019-10-24:
How you managed the tracking from 2 kinects? Can you elaborate a bit more? Than you

gitldg, 2019-10-24:
Hello @remi0s 

Now I only get the original tracking information on four computers connected with AK devices. But I came up with a solution to merge data from multiple trackers.

First of all, because AK devices are placed in different positions, each device has its own origin of coordinate system.

You can refer to my idea:
1. Correct the human body position and rotation value obtained by the tracker, and convert to a unified global coordinate system.
2. Synchronize character animation through PUN2 (found in unity asset store), and summarize skeleton data of multiple trackers
3. Finally, Compare these data(skeleton data), eliminate the abnormal data, and reconstruct the combined skeleton objects in the unified global coordinate system.

At present, I'm trying this, but I haven't finished all the functions.

closed 2019-11-11 by gitldg

Necromantic, 2019-11-21:
Let me just use this closed issue to also express my sentiment.
We worked on our own Wrapper but because of time constraints didn't have the time to work out all the kinks so far. This seems like a pretty feature complete and really well done Wrapper. So thank you for all your work and making it accessible to the public.

On the topic of the Issue I'd advice against changing the Skeleton struct for future compatibility issues. What I did was add either the index or the bodyId as a parameter for the SkeletonEvent:
SkeletonEventArgs.cs
`
    public class SkeletonEventArgs : EventArgs
    {
        public static readonly new SkeletonEventArgs Empty = new SkeletonEventArgs(-1, null);

        public SkeletonEventArgs(int bodyId, Skeleton? skeleton)
        {
            BodyId = bodyId;
            Skeleton = skeleton;
        }

        public int BodyId { get; }
        public Skeleton? Skeleton { get; }
    }`

SkeletonProvider.cs
`
        private void Update()
        {
            if (IsAvailable)
            {
                if (_tracker.TryPopResult(out var bodyFrame))
                {
                    using (bodyFrame)
                    {
                        if (bodyFrame.BodyCount > 0)
                        {
                            for (int i = 0; i < bodyFrame.BodyCount; ++i) {
                                bodyFrame.GetBodySkeleton(i, out var skeleton);
                                SkeletonUpdated?.Invoke(this, new SkeletonEventArgs(bodyFrame.GetBodyId(i), skeleton));
                            }
                        }
                        else
                        {
                            SkeletonUpdated?.Invoke(this, SkeletonEventArgs.Empty);
                        }
                    }
                }
            }
        }`