bibigone/k4a.net #19: Block(Box) Man Sample?

knat, 2019-10-26:
Thanks for the greate work of the azure kinect .net wrapper.
There's a block man sample in kinect v2, which use joint orientation to rotate the block. Could you please provide the azure kinect block man sample in MonoGame or in Unity? I find the azure kinect joint orientation value in totally different than in v2.
[image]
kinect v2 block man sample: [KinectEvolution-XAML.zip](https://github.com/bibigone/k4a.net/files/3774486/KinectEvolution-XAML.zip)

baSSiLL, 2019-11-01:
Hi @knat 
You can find some code working with joint orientations in Unity sample. Particularly, in [CharacterAnimator](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Unity/Assets/Scripts/CharacterAnimator.cs) script.

knat, 2020-02-14:
@baSSiLL Thanks, I have solved it.

closed 2020-02-14 by knat