bibigone/k4a.net #22: How to achieve point cloud？

qazqaz12378, 2019-11-07:
How can I achieve the cloud?
After I follow the #15 code, the values ​​are all 0.
[image]
[image]

baSSiLL, 2019-11-07:
You only examine a single pixel from the top-left corner (x=0, y=0). Because of [specific shape of IR illumination](https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera#illumination-mask) pixels at the corners of depth frame are always invalid and have zero depth values. Hence corresponding points of the cloud are also invalid - all zeros.
Try to examine a pixel at the center (x=depthImage.Width/2, y=depthImage.Height/2).

qazqaz12378, 2019-11-08:
This is my problem，I can already get the value correctly。
Please，How can I get the color value?

baSSiLL, 2019-11-08:
To get color value for a single depth pixel use `Calibration.Convert2DTo2D(depthCoords, depthValue, CalibrationGeometry.Depth, CalibrationGeometry.Color)`. You'll get coords of a pixel in the color frame.
If you need colors for the whole depth frame use `Transformation.ColorImageToDepthCamera`.

qazqaz12378, 2019-11-08:
I don't know how to use it.
I want to take out all the depth pixels and render the point cloud into a real color instead of a fixed color.
[image]
[image]

baSSiLL, 2019-11-08:
The last argument for `ColorImageToDepthCamera' will be filled with colors for pixels of depth frame. Thus it should have image format `ColorBgra32` and resolution of depth frame. The first argument should be raw depth frame.
You can read more about image transformations in Azure Kinect Sensor SDK here.
https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation

qazqaz12378, 2019-11-08:
[image]
[image]
According to the above, I have made the above mistake after the modification, I am already at a loss.

baSSiLL, 2019-11-08:
You allocated the array of improper size for transformed image. Bgra32 is 32-bit per pixel (1 byte per component), while in your array single pixel is short (16-bit) * 4 = 64 bits. Allocate either `int[]` array of size `depthWidth * depthHeight` or `byte[]` array of size `depthWidth * depthHeight * 4` depending on whether you need to work with individual color components or whole 32-bit color values. Also there is no need to explicitly specify a stride for Bgra32 format.

qazqaz12378, 2019-11-09:
Thank you, the problem has been solved

closed 2019-11-09 by qazqaz12378