bibigone/k4a.net #23: Cannot initialise recording to file "<<filepath>>"

viswadeep-sarangi, 2019-12-13:
I'm trying to create a Unity application that can record the captures from the Azure Kinect to a .mkv file in My Documents. I'm using the Recorder class for this. This is the statement where it throws the error:

`_recorder = new Recorder(video_output_filename, captureManager._device, captureManager.Configuration);`

Here `captureManager` is the same as that mentioned in the Unity sample project of showing the live Color and Depth stream from the Kinect to a RawTexture in Unity. 
Looking at the source code, it looks like this is the part where it throws the error:

`var res = NativeApi.RecordCreate(pathAsBytes, Sensor.Device.ToHandle(device), config, out var handle);`
`if (res != NativeCallResults.Result.Succeeded || handle == null || handle.IsInvalid)
                throw new RecordingException($"Cannot initialize recording to file \"{filePath}\".", filePath);`

Any help regarding this would really mean a lot. 
Thank you.

bibigone, 2019-12-13:
Hello @viswadeep-sarangi 

Did you try class `K4AdotNet.Record.Recorder` for recording needs?

baSSiLL, 2019-12-13:
I guess you configured format for color camera to BGRA32. Unfortunately, recording API of Kinect SDK doesn't support recording color in this format.
https://docs.microsoft.com/en-us/azure/kinect-dk/record-file-format
The most straightforward way is to set color camera format to MJPEG and use `System.Windows.Media.Imaging.JpegBitmapDecoder` to decompress frames into bitmap images for rendering.

bibigone, 2020-01-27:
In version 1.3.5 we added better diagnostic message for this case.

closed 2020-01-27 by bibigone