bibigone/k4a.net #24: K4AdotNet.Record.RecordingException: 'Error during recording to file "wpf.mkv".'

rrison, 2020-01-26:
Thanks for the Kinect DK c# wrapper.
I created a project in VS2017, and I want to record 300 frames using Recorder Class.
And I encountered RecordingException when writeCapture() is called.
below is my code snippet.
`
string filename = "wpf.mkv";

            using (var record = new Recorder(filename, device, config))
            {
                record.WriteHeader();
                button1.Text = record.FilePath;
                while (count > 0)
                {
                    record.Flush();

                    Capture capture = await Task.Run(() =>
                    {
                        return this.device.GetCapture();
                    });
                    if (capture.ColorImage != null)
                    {
                        record.WriteCapture(capture);
                        var color = capture.ColorImage;
                        TimeSpan ts = capture.DepthImage.DeviceTimestamp;
                    }
                    capture.Dispose();
                    count--;
                }
            }
`

rrison, 2020-01-26:
fixed this issue.
I put below code from init() to button_click(), it is ok now.
```c#
Device device = Device.Open();
var config = new DeviceConfiguration
{
    CameraFps = FrameRate.Thirty,
    ColorFormat = ImageFormat.ColorMjpg,
    ColorResolution = ColorResolution.R720p,
    DepthMode = DepthMode.NarrowView2x2Binned,
};
device.StartCameras(config);
```

closed 2020-01-26 by rrison

bibigone, 2020-05-28:
Just to be informed: the latest version supports recording of BGRA color format.