bibigone/k4a.net #26: About create multiple body tracks

Mediary, 2020-03-23:
I try to use multiple devices to get the same bone at the same time, but the program tells me
" Current version of Body Tracking runtime does not support creation of multiple body trackers.",
How can I solve this problem, or how to use multiple cameras to obtain bone information at the same time?
Thank you,

bibigone, 2020-05-28:
Hello @Mediary,
Please let us know what version of `K4AdotNet` library are you using? Because there was such limit before, but starting from **v1.3.5** (Jan 27) multiple body trackers are supported.

Mediary, 2020-05-29:
> 您好@Mediary，
> 请告诉我们`K4AdotNet`您使用的是哪个版本的库？因为之前有这样的限制，但是从**v1.3.5**（1月27日）开始，支持多个人体跟踪器。

> Hello @Mediary,
> Please let us know what version of `K4AdotNet` library are you using? Because there was such limit before, but starting from **v1.3.5** (Jan 27) multiple body trackers are supported.

That's the problem I encountered two months ago. I don't remember which version I used at that time. In order to solve the problem, I'm using the tools provided by other developers

Thank you for your reply

closed 2020-05-29 by bibigone