bibigone/k4a.net #29: Unity crashes on exit standalone app

ShogoTakagi, 2020-08-19:
hi, thank you for the awesome library.
I'm faced with a problem that crash (freeze) on Unity standalone app.
It seems to be related body tracking.

# How to reproduce
1. Make new scene on Unity
2. Create a game object for K4A
3. Attach "CaptureManager" and "SkeletonProvider" to the object
4. Save and Build scene (Standalone, Windows x86_64)
5. Open build folder and check DLLs (I placed on same directory of .exe)
6. Open the application and wait few seconds (Because it takes time to use body tracking)
7. Press Alt+F4 (I tried quit by coding but same result)

How can I safely quit?

bibigone, 2020-09-30:
Hello @ShogoTakagi,
Thank you for reporting and sorry for delayed reply. I cannot reproduce such crashing, but I've addressed minor improvements in the latest version that can help. Could you please give it a try.

mhaenke, 2020-10-07:
experiencing a similar issue. in the editor the program exits successfully. once compiled the program becomes non-responsive on exit.  

it seems to be a azure kinect problem
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1257

mhaenke, 2020-10-07:
experiencing a similar issue. in the editor the program exits successfully. once compiled the program becomes non-responsive on exit.

it seems to be a azure kinect problem
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1257

*i should point out am not using the latest release. just wanted to identify its not a unique issue

mhaenke, 2020-10-12:
@ShogoTakagi the latest version fixes this issue. The k4a.dll, onnxruntime.dll ... files need to be left in the .\Plugins folder (previously I had to move them to the .\build folder for the body tracking to work).

thank you @bibigone

closed 2021-04-06 by bibigone