bibigone/k4a.net #30: BodyIndexMap is unstable during multiple body tracking

mkstmyk, 2020-09-17:
Thank you for this great library. I'm enjoying development in Unity using this, but facing difficulty when capturing multiple people.
I defined specific color for each body index and paint each generated mesh for each detected person by the predefined color. When more that 2 persons are detected, the colors exchange randomly and flickeringly.
I don't observe this problem when I running the official Azure Kinect Body Tracking Viewer.
Is this known issue? Or am I doing something wrongly? 
I appreciate any comments and suggestions.
Thank you.

bibigone, 2020-09-30:
Hello @mkstmyk,
Sorry for delay with reply. We also experienced similar issue but it was not critical in our scenario.
I don't think that managed wrapper can cause such behaviour, because this wrapper is very simple and straightforward.
If you don't see the same issue in the official Azure Kinect Body Tracking Viewer, it can mean that this issue is somehow connected to multithreading (Azure Kinect Body Tracking Viewer is a single thread application, while Unity code uses multiple threads to work with Azure Sensor and Body Tracker objects).
You can also ask MS Team about it: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues

mkstmyk, 2020-11-06:
Hi @bibigone, Thank you so much for your insightful suggestions. It's embarrassing, but it seems that just converting body index to body ID solves the problem. I was mixing up them w/o reading the MS official document carefully. Sorry for asking the issue caused by myself. I hope this will help someone who are not reading the document.

closed 2020-11-06 by mkstmyk