bibigone/k4a.net #31: Playback feature throws error on standalone app

mkstmyk, 2020-12-05:
It works perfect on Editor, but once I deploy it, it shows the following error. Thank you for any suggestions in advance.

> Version 1.0.x of Body Tracking runtime is expected. stack:
  at K4AdotNet.BodyTracking.Tracker..ctor (K4AdotNet.Sensor.Calibration& calibration, K4AdotNet.BodyTracking.TrackerConfiguration config) [0x00000] in <00000000000000000000000000000000>:0 
  at MultiBodyMeshing+<>c__DisplayClass59_0.<PlaybackLoop>b__0 () [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.Tasks.Task.InnerInvoke () [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.Tasks.Task.Execute () [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.Tasks.Task.ExecutionContextCallback (System.Object obj) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.Tasks.Task.ExecuteWithThreadLocal (System.Threading.Tasks.Task& currentTaskSlot) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.Tasks.Task.ExecuteEntry (System.Boolean bPreventDoubleExecution) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.Tasks.Task.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <00000000000000000000000000000000>:0

mkstmyk, 2020-12-08:
Let me share my update. By changing the scripting backend from IL2CPP to MONO, it just worked. But I prefer IL2CPP. I hope someone knows the trick to run it w/ IL2CPP, it seems to be more general question not limited to this repo though. Thanks.

bibigone, 2021-04-06:
I am afraid that IL2CPP is not supported

closed 2021-04-06 by bibigone