bibigone/k4a.net #35: Cannot Initialize Body Tracking Runtime

tctoki, 2021-05-24:
Hi, I have some troubles running body tracker.

I downloaded the v1.4.11 source code and build it successfully. However, when I tried the WPF Body Tracker, It was saying that Cannot Initialize Body Tracking Runtime. 
I tried the Unity sample, it was also saying cannot initialize body tracking although I have put all the needed files in the Plugins/k4AdotNet folder.

In the WPF Body Tracker, it worked when the processing mode is set to CPU, but didn't work when set to GPU.

I tried the body viewer exe directly from the Azure Body tracking SDK, it works well.

I wonder how to solve this problem? I tried doing the exactly same things on another PC and it works well without error. Is there any possible reason? 

Thank you in advance!

tctoki, 2021-05-24:
I am using Azure Kinect SDK  v.1.4.1 and Body Tracking 1.1.0. My graphic card is GTX 1080.

TheWhetherMan, 2023-04-26:
I have this issue as well (a year later from the looks of it) for a Unity project. I have triple checked that all files are in the appropriate locations.

The key differences in behavior that I've tested are:
* **WORKING**: Running in Unity with all expected files in the Plugins/K4AdotNet folder
* **NOT WORKING**: Running the same Unity project, but in a standalone build

If I have the Body Tracking SDK folder in Program Files/, **all cases work fine**, but if that folder is not there then the latter case doesn't. I feel like it's not looking in the application folder(s) for the dependencies. 

I've also tried putting all of the *.dll and *.onnx next the to the .exe, as well as the aforementioned Plugins/ directory.

[image]

I can make it work in the meantime by having the SDK installed, but any ideas would be helpful for future attempts.