bibigone/k4a.net #36: capture function problem

fberna, 2021-06-03:
Hi guys,
I am using the capture function in order to create directly a .mkv file from Unity.
From the code below a get only a frame into the file. It looks like am continuously overwriting the file.
Have some of you any suggestions?

Thx
FB

> private async Task KinectLoop()
>     { 
>     while (true)
>         {
>             using (K4AdotNet.Sensor.Capture capture = await Task.Run(() => kinect.GetCapture()).ConfigureAwait(true))
>                 {
>                      using (var recorder = new Recorder(dstFilePath + "00.mkv", kinect, configuration))
>                      {
>                         if (capture != null)
>                         {
>                                 recorder.WriteHeader();
>                                 recorder.WriteCapture(capture);
>                        }
>                     }
>                 }
>         }
>     }

baSSiLL, 2021-06-03:
> It looks like am continuously overwriting the file.
Yes, that's exactly what's happening. Move createing/disposing of `recorder` and `recorder.WriteHeader` call out of the loop.
```c#
using (var recorder = new Recorder(dstFilePath + "00.mkv", kinect, configuration))
{
    recorder.WriteHeader();
    while (true)
    {
        // get capture
        recorder.WriteCapture(capture);
    }
}
```

fberna, 2021-06-03:
> Move createing/disposing of recorder and recorder.WriteHeader call out of the loop.

I tried but like that Unity crashes every time. It is probably due to RAM overload.

baSSiLL, 2021-06-03:
You should dispose `Capture` objects when finished working with them. In this case, right after calling `WriteCapture`.

fberna, 2021-06-03:
> You should dispose Capture objects when finished working with them. In this case, right after calling WriteCapture.

In this way, I will have my file in the right format but with zero frames on it.

```
using (K4AdotNet.Sensor.Capture capture = await Task.Run(() => kinect.GetCapture()).ConfigureAwait(true)
           {
                using (var recorder = new Recorder(dstFilePath + "00.mkv", kinect, configuration))
                {
                    recorder.WriteHeader();
                    while (true)
                    {
                          recorder.WriteCapture(capture);
                          capture.Dispose();
                     }
                }
            }

```

baSSiLL, 2021-06-03:
Apparently you should be getting `Capture` inside the loop before writing it to `recorder`.

fberna, 2021-06-04:
It works! Thank you

closed 2021-06-04 by fberna