bibigone/k4a.net #37: Playback function

fberna, 2021-06-10:
Hi,
I am trying to use properly the playback function in order to have a "capture variable". From this variable, I will extract color and vertice coordinates. 
My idea was just to reverse the method of recording, but it seems not to work. 
Do you have any suggestions?
Thank you
FB

```
 private async Task KinectLoop()
    {
        while (true)
        {
            using (playback = await Task.Run(() => new Playback(mkvPath)).ConfigureAwait(true))
            {
                {
                    //Initialize Kinect Configuration
                    playback.GetRecordConfiguration(out var recordConfig);
                    playback.GetCalibration(out calibration);

                    config = new DeviceConfiguration
                    {
                        CameraFps = recordConfig.CameraFps,
                        ColorFormat = recordConfig.ColorFormat,
                        ColorResolution = recordConfig.ColorResolution,
                        DepthMode = recordConfig.DepthMode,
                    };

                    playback.TryGetNextCapture(out Capture capture);
                   }
            }
       }
    }
```

baSSiLL, 2021-06-10:
Try this code
```c#
using (var playback = new Playback(mkvPath))
{
    playback.GetCalibration(out var calibration);
    playback.GetRecordConfiguration(out var config);
    while (playback.TryGetNextCapture(out var capture))
    {
        using (capture)
        {
            // use data from the capture
        }
    }
}
```

fberna, 2021-06-10:
I tried with:

```
 private void Test()
    {
        using (var playback = new Playback(mkvPath))
        {
            Debug.Log(mkvPath);
            playback.GetCalibration(out var calibration);
            playback.GetRecordConfiguration(out var config);
            while (playback.TryGetNextCapture(out var capture))
            {
                using (capture)
                {
                    Debug.Log(capture.DepthImage.WidthPixels);
                 }
            }
         }
    }
```
It gives me: "NullReferenceException: Object reference not set to an instance of an object."
The mkv path seems to be correct. There is a way to check if the capture is really open?

fberna, 2021-06-10:
You are right, TryGetNextCapture is correctly creating a valid Capture object.
I checked into MKV recorded are present 3 tracks (depth, IR, color).
[image]

Regarding the `config.DepthMode: NarrowViewUnbinned`. 

I am debugging that's why I inserted `Debug.Log(capture.DepthImage.WidthPixels);`
In reality, it looks more like this:

 ```
while (playback.TryGetNextCapture(out var capture))
            {
                using (capture)
                {
                    Debug.Log(capture.ColorImage.WidthPixels);
                    //Getting color information
                    byte[] colorArrayBuffer = new byte[capture.DepthImage.WidthPixels * capture.DepthImage.HeightPixels * 4];
                    using (var transformation = calibration.CreateTransformation())
                    {
                        using (var colorArrayImage = Image.CreateFromArray(colorArrayBuffer, ImageFormat.ColorBgra32, capture.DepthImage.WidthPixels, capture.DepthImage.HeightPixels))
                        {
                            transformation.ColorImageToDepthCamera(capture.DepthImage, capture.ColorImage, colorArrayImage);
                        }
                    }
```

baSSiLL, 2021-06-10:
So on which line the **NullReferenceException** is thrown and what's in its call stack?

fberna, 2021-06-10:
For exemple:
`Debug.Log(capture.ColorImage.WidthPixels);`

also if I try:
``Debug.Log(capture.ColorImage);``

it gives me **Null**

baSSiLL, 2021-06-10:
Here's an excerpt from the description of **TryGetNextCapture** method

> Capture objects returned by the playback API will always contain at least one
image, but may have images missing if frames were dropped in the original recording.
When calling K4AdotNet.Sensor.Capture.ColorImage, K4AdotNet.Sensor.Capture.DepthImage,
or K4AdotNet.Sensor.Capture.IRImage, the image should be checked for null.

Thus you just need to add a check that all images needed by your code are present in a **Capture** object. If not, you can either skip that capture altogether or take missing images from the previous capture.

fberna, 2021-06-17:
I tried to check as you suggest but it seems like it is always checking the same frame. After disposing of capture the increment should be automatic, right?

```
while (playback.TryGetNextCapture(out Capture capture))
            {
                using (capture)
                {
                    Debug.Log(capture.IsDisposed);

                    if (capture.DepthImage == null && capture.ColorImage == null && capture.IRImage == null)
                    {
                        Debug.Log("frame missing");
                        
                    }
                    else
                    {
                        
                        Debug.Log(capture.ColorImage.Format);
                    }

                    capture.Dispose();
                  }
             }
```

baSSiLL, 2021-06-17:
**TryGetNextCapture** advances to the next frame in a video file each time you call it.
I think you've got your check wrong. You check if ALL images are missing at once, otherwise access a color image. But your check will also pass (to the else clause) if only a color image is missing. Thus you'll get an exception in this case.
So when you use a color image check if a color image is present.

fberna, 2021-06-17:
Yes, I agree with you. It was more to check if any of the images wasn't missing. 
```
while (playback.TryGetNextCapture(out var capture) == true)
            {
                    Debug.Log(capture.IsDisposed);

                    if (capture.ColorImage == null)
                    {
                        Debug.Log("frame missing");
                        
                    }
                    else
                    {
                        Debug.Log(capture.ColorImage.Format);
                    }

                    capture.Dispose();
                    Debug.Log(capture.IsDisposed);
             }            
```

The result doesn't change. I will have all the frames that are missing. I tried also different videos.
Probably, I do something wrong with the sequence...

fberna, 2021-06-17:
In editor log I receive these error for every frames:
```
[error] [t=6076] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\allocator\allocator.c (118): k4a_capture_t_get_context(). Invalid k4a_capture_t 0000019DBDB52140
[error] [t=6076] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\allocator\allocator.c (295): Invalid argument to capture_dec_ref(). capture_handle (0000019DBDB52140) is not a valid handle of type k4a_capture_t
```

fberna, 2021-06-23:
@baSSiLL do you have any suggestions?

fberna, 2021-08-05:
Finally, I was able to have access to the images but the problem is still unknown. I opened the program once and it was working...

Right now, my problem is that running the code below Unity stacks and displays my video's last frame after some minutes.

```csharp
while (playback.TryGetNextCapture(out Capture capture))
            {
              if (capture.DepthImage != null)
                    {
                       transformation = calibration.CreateTransformation();

                        int[] xyzImageBuffer = new int[capture.DepthImage.WidthPixels * capture.DepthImage.HeightPixels * 3];
                        int xyzImageStride = capture.DepthImage.WidthPixels * sizeof(short) * 3;
                      
                        using (var transformation = calibration.CreateTransformation())
                        {
                            using (Image xyzImage = Image.CreateFromArray(xyzImageBuffer, ImageFormat.Custom, capture.DepthImage.WidthPixels, capture.DepthImage.HeightPixels, xyzImageStride))
                            {
                                transformation.DepthImageToPointCloud(capture.DepthImage, CalibrationGeometry.Depth, xyzImage);
                            }
                           
                        }

                        //num is width x height;
                        for (int i = 0; i < num; i++)
                            {

                                vertices[i].x = xyzImageBuffer[i] *      0.001f; //0.001
                                vertices[i].y = -xyzImageBuffer[i + 1] * 0.001f;
                                vertices[i].z = xyzImageBuffer[i + 2] *  0.001f;

                            i = i + 3;
                            }

                            mesh.vertices = vertices;
                            mesh.RecalculateBounds();
                            }
           }

```

closed 2021-08-13 by fberna